xlsynth/bedrock-rtl #73: RDC-robust credit initialization

bgelb-openai, 2024-10-24:
It would be useful to define a credit exchange protocol for credit/valid style channels that cross a reset domain boundary.

The baseline credit_stall based protocol is a good start, but really only handles initial reset release:

- sender drives a "stall" signal to the receiver when it is in reset
- if receiver comes out of reset first, it won't release credits to the sender until the sender deasserts its stall signal

What is needed is to handle arbitrary re-entry into reset of one or both sides of the channel. The desired behavior is:
- data can get lost (unavoidable)
- channel always reaches a consistent state where both sides have a correct view of credits (and the correct amount of credit) after either (or both) sides undergo any arbitrary reset
- either side get a positive notification when the other side goes through a reset, so that it can take any cleanup action (if necessary/desired)
- data in flight at time of reset assertion will never be delivered at the destination *after* the re-initialization completes (i.e. any data after the notification can be assumed valid from the current reset epoch)

I haven't mapped out in detail, but think this should need like a 3-4 state FSM on each side, the state of which is exported to the other link partner.

bgelb-openai, 2024-10-24:
I would not want this common component to be too opinionated about a larger reset architecture. Surely it could provide useful capabilities to that end, but I don't think that is necessarily the point.

I think the desired behavior here is good from a resilience POV generally. If two independent components are connected by a credited channel, and one of them encountered an error/problem that can only be recovered from via a reset, it is generally possible to reset just that component (perhaps in combination w/ some additional action to quiesce traffic at a higher level) and recover.

That is a good property that generally opens up a lot of options from a survivability POV.

mgottscho, 2024-11-27:
Summarizing some offline discussion. Not trying to design a general reset architecture here. The goal is to support credit/valid flow control where there can be a reset domain crossing or a reset skew boundary in between sender and receiver.

Currently the credit modules support only a reset skew boundary, i.e., both sender and receiver must enter and exit reset (with some skew in entry and exit times). We want to generalize it so that sender and receiver can independently enter and exit reset without breaking the flow control correctness, i.e., no credits should be permanently lost.

We think this can be done by changing `credit_stall` to mean `sender_in_reset` (wire going from sender to receiver) and then adding a second signal called `receiver_in_reset` (wire going from receiver to sender). These wires will control the reinitialization of credit counters on either end and also "transmission gate" the datapath as needed.

This scheme won't actually quiesce traffic or do transaction-level reset protocol. But we feel having a reset-robust flow control mechanism is a necessary (but insufficient) condition for general reset architecture.

zhemao-openai, 2024-11-27:
I would suggest we keep the name `credit_stall` and name the new signal `credit_reinit`. This will make the name match its function (reinitializing credit count at the sender) rather than its origin.

bgelb-openai, 2024-11-27:
credit_dump?

zhemao-openai, 2024-11-27:
The name on the regular counter is `reinit`, so I think `credit_reinit` would be most consistent.

zhemao-openai, 2024-12-11:
Took a stab at it in #244 . The signals are named credit_reset_active_fwd (sender to receiver) and credit_reset_active_rev (receiver to sender). When the sender gets credit_reset_active_rev, it reinitializes its credit counter and blocks sending valid. When the receiver gets credit_reset_active_fwd, it reinitializes its credit counter and stalls sending any credit. That should be sufficient to allow either side to be reset without the other.

bgelb-openai, 2025-01-10:
Pasting @mgottscho summary from slack:

> Yeah right now we only have push_credit_stall. The original design intent was that sender and receiver would be in the same reset domain (though potentially with reset skew, hence needing the stall signal). This design is safe, but not as convenient as it could be.
We agreed it would be more convenient if the design was generalized so that one can place the reset domain crossing in between the sender and receiver if they want to (i.e., a common case being at a credited FP block boundary). To make that work, we needed a two-signal protocol rather than one-signal.
So push_credit_stall will be renamed to sender_in_reset and we will add a second wire called receiver_in_reset as 
[@bgelb](https://openai-corpws.slack.com/team/U073TPKQK8R)
 described above. (This change has not yet been implemented.)
The credit libs will make sure that the credit counts are eventually consistent across sender and receiver when either or both of them get reset. It will guarantee credits don't get lost and that the channel can come alive again and flow control normally. It will not implement anything more general than that, i.e., it's not some general-purpose reset handshake mechanism.
[@zhemao](https://openai-corpws.slack.com/team/U07UE0D3RK3)
 had taken a stab at this change in December but I think the PR needs some updates.

mgottscho, 2025-01-10:
might need to scrap #244 and start over

PR #323 (2025-01-14): Two-wire handshake for resetting credit sender & receiver + fix "in reset" assert macros
Fixes #73 and supersedes #244
commit 2bdd8a1e (2025-01-14): Two-wire handshake for resetting credit sender & receiver + fix "in reset" assert macros (#323)

Fixes #73 and supersedes #244
closed 2025-01-14 by mgottscho