xlsynth/bedrock-rtl #130: Find way to guarantee that assertion failures cause simulation to exit with nonzero exit code

zhemao-openai, 2024-11-12:
We are currently setting the minimum assertion failures to 1 so that the simulation aborts on the first failure. However, it would be helpful to let the simulation run to completion and then exit with failure if any assertions were triggered. It seems this may not be the case currently if the minimum assertion failures is relaxed. Find a way to do this properly in the test driver.

mgottscho, 2024-12-17:
I'm curious why you find this useful. If an assertion fires and it's legitimate, then the behavior is already undefined, so any downstream waves and events would be suspect, no?

mgottscho, 2025-03-19:
Closing this issue, because VCS and dsim should both exit nonzero if assertions happen. Additionally, our verilog runner plugins search for assertion macro text in the logs.