xlsynth/bedrock-rtl #159: Implement loadable shift register

zhemao-openai, 2024-11-15:
We have the delay modules that can take serial input. What's missing is a shift register module that can be loaded in parallel. I imagine an interface something like this:

```
module br_shift_reg #(
  parameter int BitWidth = 1,
  parameter int NumStages = 2
) (
  input. logic clk,
  input. logic rst,

  input. logic reinit,
  // value gets set to initial_value on reset or when reinit is asserted
  input  logic [NumStages-1:0][BitWidth-1:0] initial_value,
  output logic [NumStages-1:0][BitWidth-1:0] value,

  input. logic shift_en,
  input  logic [BitWidth-1:0] shift_in,
  output logic [BitWidth-1:0] shift_out
);
```

mgottscho, 2024-11-15:
lgtm

zhemao-openai, 2024-11-23:
This was fixed by PR#189

closed 2024-11-23 by zhemao-openai