xlsynth/bedrock-rtl #303: br_fifo_ctrl_1r1w_push_credit.sv returns credit during reset

bgelb-openai, 2025-01-10:
if you use br_fifo_ctrl_1r1w_push_credit with RegisterPushCredit = 0 it will incorrectly drive push_credit high under the following condition:

rst = 1 (in reset!)
push_credit_stall = 0

Since the other end of the connection does not observe the local reset, it has no way to tell the "credit" being sent over the push_credit wire is invalid, and receives an excess of credits.

I believe even when not flopped, the push_credit output should be 0 if rst input is asserted.

bgelb-openai, 2025-01-10:
Please let me know what you think mark. I could probably make a stab at fixing if you agree w/ my diagnosis.

cc: @taoliu-o

mgottscho, 2025-01-10:
I agree with your analysis. The registered case is WAI because the `push_credit` reg does reset to 0:

```
  if (RegisterPushCredit) begin : gen_reg_push
    `BR_REG(push_credit, push_credit_internal)
  end else begin : gen_passthru_push
    assign push_credit = push_credit_internal;
  end
```

If you'd like to go ahead and send a fix please do! :)

mgottscho, 2025-01-10:
I think it is just

```
assign push_credit = !rst && push_credit_internal;
```

bgelb-openai, 2025-01-10:
is it ok to have rst->out combo path? was thinking of sticking a flop.

PR #304 (2025-01-10): dont allow push_credit to assert during reset
Fixes #303 

cc: @masai-openai @taoliu-o @zhemao-openai
commit 526b94de (2025-01-10): dont allow push_credit to assert during reset (#304)

Fixes #303 

cc: @masai-openai @taoliu-o @zhemao-openai
closed 2025-01-10 by bgelb-openai