xlsynth/bedrock-rtl #667: Tweak ready/valid protocol checks

obowen-openai, 2025-05-01:
The current ready/valid protocol checks don't consistently check for data being known when valid. That check only happens when the data stability requirement is waived. It is also not possible to disable the known-data check without also disabling the valid-stability check.

Current implementation:

- valid stability check enabled && data stability check enabled --> check valid and data bits are stable, but do not check data bits are known when valid
- valid stability check enabled && data stability check disabled --> check valid is stable and check data bits are known when valid
- valid stability check disabled --> we do not check data bits are known when valid

Proposed implementation:
- Always check data bits are known when valid, regardless of stability checks, and add a separate parameter to disable that check
- valid stability check enabled && data stability check enabled --> check valid and data bits are stable, don't check data is known
- valid stability check enabled && data stability check disabled --> check valid is stable, don't check data is known

mgottscho, 2025-05-02:
We had written it this way because we assumed `$stable(1'bx)` would always be false. It turns out this is not necessarily the case. LRM is ambiguous:
> $stable returns true (1'b1) if the value of the expression did not change. Otherwise, it returns false (1'b0).

Agree with the proposed implementation.

obowen-openai, 2025-05-02:
Just one minor point is that the current implementation only checks data stability during a backpressure event (when EnableAssertDataStability is high), so even if stable(1'bx) were to return false, it wouldn't catch the case where you have unknown data during a non-backpressured transaction.