xlsynth/bedrock-rtl #668: Make br_flow_deserializer's internal flopped data usable

mattskl-openai, 2025-05-02:
https://github.com/xlsynth/bedrock-rtl/blob/668ccc3c4edf91ddabc74b0a891b134088e4f74d/flow/rtl/br_flow_deserializer.sv#L246-L251

I see that the deserializer is optimized for minimizing storage space, passing through the last flit directly to the output.
I was planning to use the deserializer internal storage as an entry flop..because that's what it is (except for the last flit).

Can we make this an option?
As part of this, push_ready should be high for last flit even if !pop_ready, and the last flit push will go to the extra flop storage.
And pop data can select from current push or the flop storage.

obowen-openai, 2025-05-02:
If you don't need the cut-through latency of zero, you can just have a br_flow_reg_fwd + br_flow_deserializer. If you do need cut-through latency of zero, but can tolerate 1 cycle of backpressure relief latency, you can use br_flow_reg_rev + br_flow_deserializer. If however you do need a cut-through latency of zero and no latency on backpressure relief, then we could imagine having a new `br_flow_reg_bypass` module which acts as a 1 deep fifo with bypass and combinational paths from pop_ready to push_ready. I think that might be a useful building block to have in the library anyway and may be a clean way to address this issue.

obowen-openai, 2025-05-02:
If you add a narrow 64b flow reg just before the deserializer you've effectively created a 1-deep 2048-wide buffer slot plus deserializer, using 2048 total flops.

closed 2025-05-02 by mattskl-openai

obowen-openai, 2025-05-02:
Okay, so you just want the convenience of having it combine them for you.

FWIW, I would say the deserializer has one job to do, and it doesn't need to do anything else. If we were to bake in a buffer stage, then we need to add options for what type of buffer stage to add since there are various tradeoffs you can make with latency and timing. So IMO, it's probably best to just provide the building blocks in the lib and let the user stitch things together as they need and/or provide whatever convenience wrappers they want for their project.