xlsynth/bedrock-rtl #725: `br_arb_grant_hold` footgun

mgottscho, 2025-05-22:
Currently, if `enable_priority_update` is 0, you can still set `grant_hold` to 1, which is probably not what the user intends.

We should constrain the design a bit more:
* `enable_priority_update == 0` --> don't hold grant, and don't let arb update
* `enable_priority_update == 1` --> allow grant to held, and let arb update either way

cc @bgelb-openai @ajaffey-openai @zhemao-openai

mgottscho, 2025-05-22:
I think for the grant holder, we should also rename `enable_priority_update` to `accept` to better convey the behavior, since there would still be an output port to the arb called `enable_priority_update`

mgottscho, 2025-05-22:
actually i'm not certain this is the ideal fix. i think we should have one control bit, not two

mgottscho, 2025-05-22:
@bgelb-openai please clarify use cases

bgelb-openai, 2025-05-23:
On a just a regular arbiter (not grant/hold) we have the enable_priority_update input that can be used to prevent the arbiter state from updating on a grant.

Why do we have this? We can use it in case a requester wins arbitration, but then the arbitrated resource itself is unavailable (e.g. the channel is backpressuring, or a we fail to win arbitration in a second stage). In particular, it is needed when the downstream availability depends on which requester is granted (this case shows up in a NOC router, where we have input arbiters over VCs, and output arbiters over input ports). The "enable_priority_update" input allows to find out who gets granted but then not "accept" the grant if it turns out the transfer doesn't happen.

So the thinking was to extend this to the grant/hold circuit as well. My thought at least was to generalize it to multi-cycle arbitration via the grant/hold circuit. If the resource being arbitrated for turns out to be unavailable, you can cause the grant/hold circuit to not update its state.

Though there is an enable_priority_update port on the grant/hold, it doesn't actually prevent the grant/hold state from changing. Which seems potentially problematic. Its also not very useful for anything as far as I can see.

I suppose perhaps its not quite as necessary as the port on the base arbiter itself, in the sense that you can still "dry run" the arbiter itself w/o accepting the grant (updating the state). But feels like it would make more sense.

Without the use of this port, I had to do the following instead in a case where we have AXI traffic (the final beat of the transfer has a LAST bit).

assign request = input_valid;
assign grant_hold = ~(input_valid & input_last);

// "gating" to keep the grant/hold state from changing
assign gated_request = request & {N{output_ready}};
assign gated_grant_hold = output_ready ? grant_hold : grant;

When the output channel backpressures, we force all request inputs to 0 and feed the grant back to the grant hold to preserve the hold. This works in the particular place I was trying to use it. But it seems like it might be more natural to let the enable_update port to integrate this function?

I haven't thought hard about what would happen in a multi-stage allocator like that NOC router case I mentioned. Might be a little more awkward, because I need to get the grant but then might not accept it, so some additional side-plumbing might be needed to get that to work.

mgottscho, 2025-07-22:
@bgelb-openai I think we are done with this? Can we close it?